andreiva/simple-rssi-antenna-tracker #1: error: 'Timer' does not name a type

MoistNugget666, 2017-01-08:
`simple-rssi-antenna-tracker.ino:101: error: 'Timer' does not name a type
simple-rssi-antenna-tracker.ino: In function 'void setup()':
simple-rssi-antenna-tracker.ino:122: error: 'timer' was not declared in this scope
simple-rssi-antenna-tracker.ino: In function 'void loop()':
simple-rssi-antenna-tracker.ino:129: error: 'timer' was not declared in this scope
`
hello,
im fairly new to adruino programming and get this when i try to compile. any idea how to fix it?

andreiva, 2017-01-09:
Hi,
the problem is missing Timer.h library, you need to install it manually.
Download ZIP file from here https://github.com/JChristensen/Timer  then
extract zip content in (assuming youre using windows) My
Documents\Arduino\libraries\

There are several ways to install libraries, check out this video:
https://www.youtube.com/watch?v=K1cIWCBKaUY

-Andrei

yoowicked, 2018-12-06:
Need help here! I am using arduino1.0.5 and have installed the timer library but still having the same error message. Any advice to sort it out?

yoowicked, 2018-12-06:
By the way, the video link in your comment is no longer available. Do you have another link? Thanks!

andreiva, 2018-12-11:
Hi, just search YouTube "how to install libraries in adruino" or something
similar. There should plenty of results.