andreiva/simple-rssi-antenna-tracker #3: Cant compile

FranzAr, 2017-10-02:
Tried to compile the sketch, but won't work. Any idea what may be the problem

Greets, Franz

```
Arduino: 1.8.5 (Windows 8.1), Board: "Arduino Nano, ATmega328P"

Build-Optionen wurden verändert, alles wird neu kompiliert
D:\arduino\tracker\simple-rssi-antenna-tracker\simple-rssi-antenna-tracker.ino: In function 'void setup()':

simple-rssi-antenna-tracker:122: error: no matching function for call to 'Timer::every(int, void (&)())'

   timer.every(50, mainLoop);

                           ^

D:\arduino\tracker\simple-rssi-antenna-tracker\simple-rssi-antenna-tracker.ino:122:27: note: candidates are:

In file included from D:\arduino\tracker\simple-rssi-antenna-tracker\simple-rssi-antenna-tracker.ino:10:0:

C:\Users\Franz\Documents\Arduino\libraries\Timer-2.1/Timer.h:40:10: note: int8_t Timer::every(long unsigned int, void (*)(void*), void*)

   int8_t every(unsigned long period, void (*callback)(void*), void* context);

          ^

C:\Users\Franz\Documents\Arduino\libraries\Timer-2.1/Timer.h:40:10: note:   candidate expects 3 arguments, 2 provided

C:\Users\Franz\Documents\Arduino\libraries\Timer-2.1/Timer.h:41:10: note: int8_t Timer::every(long unsigned int, void (*)(void*), int, void*)

   int8_t every(unsigned long period, void (*callback)(void*), int repeatCount, void* context);

          ^

C:\Users\Franz\Documents\Arduino\libraries\Timer-2.1/Timer.h:41:10: note:   candidate expects 4 arguments, 2 provided

simple-rssi-antenna-tracker:123: error: no matching function for call to 'Timer::every(int, void (&)())'

   timer.every(5, measureRSSI);

                             ^

D:\arduino\tracker\simple-rssi-antenna-tracker\simple-rssi-antenna-tracker.ino:123:29: note: candidates are:

In file included from D:\arduino\tracker\simple-rssi-antenna-tracker\simple-rssi-antenna-tracker.ino:10:0:

C:\Users\Franz\Documents\Arduino\libraries\Timer-2.1/Timer.h:40:10: note: int8_t Timer::every(long unsigned int, void (*)(void*), void*)

   int8_t every(unsigned long period, void (*callback)(void*), void* context);

          ^

C:\Users\Franz\Documents\Arduino\libraries\Timer-2.1/Timer.h:40:10: note:   candidate expects 3 arguments, 2 provided

C:\Users\Franz\Documents\Arduino\libraries\Timer-2.1/Timer.h:41:10: note: int8_t Timer::every(long unsigned int, void (*)(void*), int, void*)

   int8_t every(unsigned long period, void (*callback)(void*), int repeatCount, void* context);

          ^

C:\Users\Franz\Documents\Arduino\libraries\Timer-2.1/Timer.h:41:10: note:   candidate expects 4 arguments, 2 provided

exit status 1
no matching function for call to 'Timer::every(int, void (&)())'

```

andreiva, 2017-10-03:
Looks like you haven't installed Timer library. Take a look at README, there you'll find instructions how to do it.

closed 2017-10-03 by andreiva

FranzAr, 2017-10-03:
I'm sorry but timer is installed. Tried it on 2 different machines and several Arduino versions. Can't compile it, alway the same errors. Maybe I'm to stupid, sorry

FranzAr, 2017-10-03:
Found it, 2.x timer lib is not working, had to take old 1.3 branch. Now it works. Maybe you should mention this somewhere ;-) Thank you

andreiva, 2017-10-03:
Okay thanks, I'll check it out later.