andreiva/simple-rssi-antenna-tracker #4: code issue

vannoz, 2017-11-23:
hello Andrei, I got an error when trying to compile this code...
[code]
`//#include <LiquidCrystal.h>

/*
  Simple RSSI Antenna Tracker
  RELATIVE
*/

#include <Servo.h>
#include <Timer.h>

/*
    For testing different curves
    All new curves are more precise
    and maintain lock better than old RELATIVE

    Uncomment one
*/
//#define EXPONENTIAL       // OK
//#define RELATIVE          // old, don't use
#define SIGMOID             // Best
//#define PROPORTIONAL      // twitchy

#define SIGMOID_SLOPE       1
#define SIGMOID_OFFSET      4

#if !defined(EXPONENTIAL) ^ !defined(SIGMOID) ^ !defined(PROPORTIONAL) ^ !defined(RELATIVE)
// all good
#else
#error "Please define ONE tracking curve: EXPONENTIAL, SIGMOID, PROPORTIONAL, RELATIVE"
#endif

/*
   Pin mapping
   You can change these pins, just make sure that
   RSSI pins are analog inputs
   and servo pin supports PWM
*/
#define LEFT_RSSI_PIN       A0     // analog RSSI measurement pin
#define RIGHT_RSSI_PIN     A1
#define PAN_SERVO_PIN    13     // Pan servo pin

/*
   There is going to be some difference in receivers,
   one is going to be less sensitive than the other.
   It will result in slight offset when tracker is centered on target

   Find value that evens out RSSI
*/
#define RSSI_OFFSET_RIGHT   0
#define RSSI_OFFSET_LEFT    0

/*
   MIN and MAX RSSI values corresponding to 0% and 100% from receiver
   See serial monitor
*/
#define RSSI_MAX            400
#define RSSI_MIN            120

/*
   Safety limits for servo

   Find values where servo doesn't buzz at full deflection
*/
#define SERVO_MAX           180
#define SERVO_MIN           0

/*
   Servo 'speed' limits
[... 299 lines cut ...]
libraries\Timer\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::oscillate(unsigned char, unsigned long, unsigned char)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::pulse(unsigned char, unsigned long, unsigned char)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timer\Timer.cpp.o (symbol from plugin): In function `Timer::Timer()':

(.text+0x0): multiple definition of `Timer::pulseImmediate(unsigned char, unsigned long, unsigned char)'

sketch\Timer.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
*********************************************************************************
thanks
best regards
Andrea

andreiva, 2017-11-28:
Were you able to solve the problem?

vannoz, 2017-11-28:
Hello...i tried to merge two .ino code, one is your code and one the
diversity rx5808 ino code....but they are using a same name library...the
timer.h
So there are some problem...can You send me your mail? I'll send you what i
did for your help...
Thanks

vannoz, 2017-11-28:
you program doesn't allow to choose the video channel, so I found this code
which is a diversity based on the RX5808:
https://github.com/sheaivey/rx5808-pro-diversity

so I tried to put them together but I find a conflict on two libraries that
have the same name (timer.h) .... and i do not have the skills to make it
one

in the attached file, I tried to make the merge ... I also entered the
commands for the TILT servo .... but putting it on arduino, the servo
trembles ... it goes back and forth as crazy..

 hardware part:
in practice the antenna will consist of 4 helical antennas: Right and Left
to control the PAN servo modified to have 360deg, Up and Down for the
TILT....
then to get the video signal i think to use a fifth helicoidal antenna
mounted on the center and connected to my ImmersionRC diversity
receiver...where the second antenna is a omnidirectional

I have also purchased a 6-wire slip ring, in practice are sliding contact
lifts, so the pan can rotate 360 ​​to infinity without wavering the cables
..

thanks
ciao
Andrea

2017-11-28 13:04 GMT+01:00 andrea vanni <andrea.vanni1@gmail.com>:

Mattxjax45, 2018-09-23:
did this ever work?