andreiva/simple-rssi-antenna-tracker #8: 360 servo support

danielolsman, 2020-05-17:
I'm trying to get this tracker software to work with a 360 degrees servo, no luck so far.
the servo only spins faster and faster.

can 360 servo support please be added if possible ?

andreiva, 2020-05-17:
Long story short, 360 is supported, but you need right servo
*"The GWS S125 1T 2BB Sail Winch Servo is a unique servo that has a wider
rotation range than typical hobby servos. It rotates one full turn (360°)
over the standard 1-2 ms servo pulse range output"*

There are two types of servos:
- continuous rotation servos (like this one
<https://hobbyking.com/en_us/towerpro-mg90d-mini-digital-servo-2-4kg-0-08sec-13g.html?queryID=041d338a3bdf9ad7756fa1cba7094be1&objectID=304&indexName=hbk_live_magento_en_us_products>)
which have no limits, they just rotate in either direction at adjustable
speed and never stop
- extended range servos (GWS S125 1T 2BB
<https://www.pololu.com/product/522>) which is what you want. This servo
goes from 0° to 360° and stops

I'm guessing you are using a continuous rotation servo?

danielolsman, 2020-05-17:
I understand now, I have a SPT 5525 servo which indeed is a continuous rotation servo.

Thanks for the explanation